nes-lab/shepherd #9: add support for WSL2

fedy0, 2022-08-12:
I prepared and tested a script to setup the latest WSL (WSL2) and [usbipd](https://github.com/dorssel/usbipd-win) services but I want to know if its needed for easy-of-test of the gcc-port on Windows.

IMHO, I think it would be helpful and better than a traditional virtual machine as WSL2 comes with kernels and one can easily swap between its version and that of the kernels.

I currently don't know how the firmware is deployed (not yet familiar with ansible) but I added usbipd just incase if we need usb sharing services between Windows and WSL2, all in a single click.

Lastly, if I am to add the script and its README, what location is preferred? `./software/gcc-port` and `./docs`  or both in `./software/gcc-port` or add none?

orgua, 2022-08-13:
Well first, the usual starting-point is having a running linux at hand. There are sufficient resources online for installing WSL if you want to use windows. It is not supported by us, as stated [here](https://github.com/orgua/shepherd/blob/master/docs/user/getting_started.rst). Additionally WSL-Routines still keep changing, last time with WSL2.

Second, what would a user need usbipd for? The only thing I can think of is your special use-case. Or did i miss something? I didn't try this yet, but is usb-sharing really needed? I thought there will be a com-port for ssh. 
Did you manage to fully setup a beaglebone connected via USB? How about providing Internet-Access and running ptp / ntp? It looks like this is too niche to include in the repo. The usual and preferred way is to include the BBs into your LAN. 

thirdly: yes, WSL is far more helpful and usable than a VM :)

at last but not least: usage of shell-scripts is somehow outdated - there are more secure and better documented alternatives available. Shepherd uses ansible. The few steps for bootstrapping the process (beforehand) are documented in readme-files. 

So no, integration of the script into the repo is unlikely, but you are welcome to post your script here in this issue - so others can find it. 

Just for completion: WSL is not needed in windows. you can ssh from a normal shell, same goes for python / ansible. And linux-bound cross-compilation could be run on the BB itself.

fedy0, 2022-08-24:
Ok,
I just added the WSL2 and usbipd script [here](https://gist.github.com/fedy0/09f7edd64a5176ad834f095f27e27864)

I hope it helps someone.

closed 2022-09-05 by orgua