nes-lab/shepherd #10: use crosstool-ng & regenerate

fedy0, 2022-09-11:
@orgua I have been thinking about regenerating the toolchain we've been using to see if some of the errors ([here](https://github.com/orgua/shepherd/blob/gcc-port/software/gcc-port/firmware/README.md)  and [there](https://github.com/orgua/shepherd/blob/gcc-port/software/gcc-port/firmware/readme_overflow_issue.md)) that seem internal to the compiler will reduce.
I've little experience with crosstool-ng.
Do I try it?
I'm asking before hand because it takes a while to generate the toolchain.
What do you think?

orgua, 2022-09-12:
Just for reference: you are talking about the c-code for the PRUs on the beaglebone.

We are currently using two different cross-compilers (TI's CGT and GCC with PRU-Extension) for the PRU-Code. One compiler (cgt) has trouble calculating correct results with u64, the other currently produces non-optimal code and therefore overflows the program memory. Additionally we use a basic makefile, comparable to a bash-script, to feed the compiler. That's our current "chain of tools".

If you know a way to get rid of these constraints go for it. It should be a good exercise to learn more about embedded software development.

orgua, 2022-10-26:
ok, closing this. CGT works for now and GCC saw big improvements [lateley ](https://github.com/dinuxbg/gnupru/issues/43#issuecomment-1272528295)

closed 2022-10-26 by orgua